psychogenic/kicad-skip #1: Adding a new symbol from the library

Monacrylic, 2024-03-13:
Thank you for the pioneering work!

I was trying add a new symbol from the KiCAD library (installed on my PC). There doesn't seem to be function that allows me to do so, or maybe I'm not using it correctly :)

psychogenic, 2024-03-13:
Hello Siddharth,

Thanks :)  

Short version: no, you're not using it wrong, there simply is no such functionality (at least at this stage).

Symbols may be clone()d, but constructing them from scratch was a toughie.   If you read in any schematic, choose one of the symbols and use the not-really-documented raw() method
`myschem.symbol[0].raw()`
you'll see, under the hood the symbols are rather complex beasts.  They are based on one of the lib_symbol items (that's only present in the schem file when that symbol type is actually in the schematic), and then derive their pins and other things from that, plus having all the properties etc.
In order to support just creating symbols, we'd have to have a way to either replicate kicad's functionality (meaning locate and parse the various symbol libraries) and then fill in all the extra symbol bits... I didn't want to start re-implementing kicad to that extent, so... it's just not there.
Maybe there's a way to do this simply, but for the moment if you want a power symbol, you'd have to

1. have another of same type in schem
2. find it--the power symbols have autogenerated names PWR1234 or something, so you actually only get to know which it is based on properties
3. clone it

Simpler objects can be created fresh, and these are actually instantiated based on templates in skip.element_template.  In theory, the same might be doable for symbols, but the raw() dump above will give you an idea of just how many elements you'd have to tweak (meaning either make the user responsible for, or actually force as arguments to the constructor).

Sorry I can't give you a better answer on that front.

Monacrylic, 2024-03-13:
Thanks Pat,

I totally understand. I was just going through the templates in skip.element_template, and understood what maybe limiting the generalization. Just wondering if there could be some code that we could borrow from skidl (https://github.com/devbisme/skidl).
Since they've already implemented the Symbol parsing from KiCAD's libraries.

That said, I'm new to library development, but I'll be super happy to take a shot at implementing this functionality. This library has the potential to open up a new world of possibilities for integrating AI into KiCAD. The lack of an eeschema API is the primary roadblock!