psychogenic/kicad-skip #2: KiCAD write issue

thesrinivasprabhu, 2024-04-04:
[image]
[image]

We manually made the resistor 50 which we are able to read from python importing your library. However when we tried to set the resistor value to 10K (line no 11) it is not reflecting back in the schematic.

PS:

- If python has been run keeping schematic open will it modify the schematic or should it be closed while running python script?

connectfevino, 2024-04-12:
i also observed the same behavior , i using kicad 8.0.1 latest version

semiversus, 2024-06-21:
When you say you don't see it reflecting back - do you mean you have kicad open and don't see a change? This library is not communicating with KiCAD, it stores it directly to the file. When you have KiCAD open, it will not check for file changes. Best way is do have KiCAD closed while working with this library and then opening it up and see the result.