psychogenic/kicad-skip #8: How to make a symbol property hidden?

bikeNomad, 2024-08-13:
If I want to make a symbol's property hidden (i.e. add a `hide = True` effect), how can I do it?

If there's already a `hide` effect I can change its value, but un-hidden properties typically don't have a `hide = False` effect.

StefanHol, 2024-11-23:
Hi psychogenic,

this is a great project. I have been able to run it to add and update properties from a SQLite database to my schematics, but the new properties are all visible and hiding them is unfortunately not possible at the moment.
Or I can't do it.

Best regards
Stefan

imagidan, 2025-08-03:
This seems to work

```
from skip import Schematic
from sexpdata import Symbol

sch = Schematic("original.kicad_sch")

for sym in sch.symbol:
    try:
        field = sym.property.PROPERTY_NAME
        effects = field.effects
        effects._tree.append([Symbol('hide'), Symbol('yes')])
        effects._nameCounts['hide'] = 1
    except AttributeError:
        continue

sch.write("updated.kicad_sch")
```