psychogenic/kicad-skip #20: The write() method of Schematic fails with an error in sexpdata.tosexp.

RobertPHeller, 2025-05-07:
I am getting this error when I try to write a schematic file:

```
 File "/home/heller/.local/local/lib/python3.11/dist-packages/sexpdata.py", line 378, in tosexp
    raise TypeError(

TypeError: Object of type '<class 'skip.sexp.parser.ParsedValue'>' cannot be converted by `tosexp`. It's value is 'value = +5V'

```
sexp.parser.ParsedValue needs a __to_lisp_as__() method.

I am using skip 0.2.5 and sexpdata-1.0.2, install via pip to a local prefix:

```
marchhare% dir ~/.local/local/lib/python3.11/dist-packages/
kicad_skip-0.2.5.dist-info/ sexpdata.py 
__pycache__/                skip/
sexpdata-1.0.2.dist-info/   

```
And kicad 6.0.11:

```
marchhare% dpkg-query -l \*kicad\*|grep ^ii
ii  kicad            6.0.11+dfsg-1 arm64        Electronic schematic and PCB design software
ii  kicad-demos      6.0.11+dfsg-1 all          Demo projects for kicad
ii  kicad-doc-en     6.0.11+dfsg-1 all          Kicad help files (English)
ii  kicad-footprints 6.0.11-1      all          Footprint symbols for KiCad's Pcbnew
ii  kicad-libraries  6.0.11+dfsg-1 all          Virtual package providing common used libraries by kicad
ii  kicad-packages3d 6.0.10-1      all          3D models for 3D viewer in KiCad's Pcbnew and Footprint Editor
ii  kicad-symbols    6.0.10-1      all          Schematic symbols for KiCad's Eeschema
ii  kicad-templates  6.0.8-1       all          Project templates for KiCad

```

(On a Raspberry Pi 5 with Debian 12)

RobertPHeller, 2025-05-08:
Yes, adding this method to ParsedValue in parser.py solves the problem:

```
    def __to_lisp_as__(self):                                                   
        return ParsedValue.toString(self)               
```

paul356, 2025-09-14:
I met the same issue when I try to add some property to some symbol. The added property is new constructed ParsedValue. I wonder why we don't hit such issue when I write out schematic without extending a property. The other intact items inside the same property are also ParsedValue's.

In my case I found it is better to change __to_lisp_as__ to:

    def __to_lisp_as__(self):
        return sexpdata.dumps(self._tree)